tensorflow/tflite-micro #3098: How to reduce time taken by Interpreter.invoke()

SachinEngibrains, 2025-04-18:
I my Python project i am using tflite_micro when I run Interpreter.invoke() ot takes ~30 mins.

Is there any way to reduce it using threading, GPU, or something else?

Please help to resolve this issue.

ddavis-2015, 2025-04-18:
@SachinEngibrains 30 minutes does seem like a long time.  How large is your model?

TFLM does not have optimized kernels for the x86 target.  Perhaps TfLite would be more suitable for your model?  TfLite does have optimized x86 kernels.  An example of how to use the Tflite interpreter can be found [here](https://www.tensorflow.org/api_docs/python/tf/lite/Interpreter).

Any issues with the TfLite interpreter should be created in the [Tensorflow repository](https://github.com/tensorflow/tensorflow/issues).

closed 2025-04-18 by ddavis-2015

SachinEngibrains, 2025-04-21:
@ddavis-2015 

Thanks for your response,

Is there any way to run TFML Invoke in multiple threads?

ddavis-2015, 2025-04-21:
> [@ddavis-2015](https://github.com/ddavis-2015)
> 
> Thanks for your response,
> 
> Is there any way to run TFML Invoke in multiple threads?

@SachinEngibrains 

No.