tensorflow/tflite-micro #3130: TFLM output does not match TfLite output for Conv OP

AIWintermuteAI, 2025-07-02:
First found during manual testing and confirmed with layer_by_layer_debugger script.
Output:
```
AssertionError: 
Arrays are not equal

TFLM output does not match TfLite output.
 Subgraph number is 0 
 Layer number is 2 
 The Tensor Index where this output does not match is 46 

Mismatched elements: 80 / 51200 (0.156%)
Max absolute difference among violations: 1
Max relative difference among violations: 1.
 ACTUAL: array([[[[   7,   14,    6,   17,    4,   -5,  -20,    8],
         [   1,  -34,   15,  -12,   15,   15,    7,    6],
         [   2,   15,   -5,   47,    6,   -6,   -3,    0],...
 DESIRED: array([[[[   7,   14,    6,   17,    4,   -5,  -20,    8],
         [   1,  -34,   15,  -12,   15,   15,    7,    6],
         [   2,   15,   -5,   47,    6,   -6,   -3,    0],...
```
The model in question is attached:

[tflite_learn_977.tflite.zip](https://github.com/user-attachments/files/21017035/tflite_learn_977.tflite.zip)

The difference is between optimized Conv kernel used in full tflite and reference kernel used in tflite micro.

Potentially related:
https://github.com/tensorflow/tflite-micro/issues/2629

ddavis-2015, 2025-07-08:
[internal: to be investigated]

dansitu, 2025-08-15:
Any further information on this issue?

ddavis-2015, 2025-08-16:
@AIWintermuteAI @dansitu @ArmRyan @Unbinilium

There is no difference in CONV kernel outputs when both TfLite and TFLM are using reference kernels.

This turns out to be an issue with comparing the outputs of TfLite optimized kernels against the outputs of reference kernels.  The TFLM CONV kernel uses the reference CONV operator [here](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/kernels/internal/reference/integer_ops/conv.h).  This reference operator is imported into TFLM from the TfLite source code, and is thus identical to TfLite.

However, the default TfLite interpreter uses an optimized CONV kernel which in turn uses [this](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/kernels/internal/optimized/integer_ops/conv.h) CONV operator.  As you can see, these are very different code paths.  The result is an off-by-one difference among some elements of the output tensor.  As to why there is a difference in the TfLite reference and optimized CONV operator outputs, you would need to ask that question in the [TfLite repository](https://github.com/tensorflow/tensorflow/issues), as a new issue.

Here is a colab example showing TfLite reference vs. optimized kernel vs. TFLM reference kernel for the CONV operator: [conv_mismatch](https://colab.research.google.com/drive/1kAU9oISAYseN-8N-r5jsax1T4TEPSH74?usp=sharing)

This issue will be resolved in a future PR for the TFLM layer-by-layer debugger, such that the default OpResolver for TfLite uses reference kernels.

ddavis-2015, 2025-08-16:
[internal: add additional command line options to the layer-by-layer debugger, allowing setting of the arena size, and which TfLite OpResolver to use]

veblush, 2025-11-14:
While working on a separate task, I had the opportunity to investigate this issue and identified the root cause. It stems from a subtle difference in the requantization algorithms used in TFLite/TFLM reference kernels compared to other implementations. I have filed https://github.com/tensorflow/tflite-micro/issues/3252 to track this for TFLM.

It is worth noting that off-by-one errors are often unavoidable, as optimized kernels frequently sacrifice absolute precision for speed. While TFLM strives for bit-exactness, TFLite does not. Consequently, you may see slightly different results when switching between TFLite Reference, TFLite Optimized, and TFLite with XNNPack (the current default), as each employs slightly different requantization logic. (I guess this requantization won't be the only one having this kind of discrepency)

> This does not seem to be wise, if optimized tflite kernels are used by default. The only thing that will achieve is more confusion, when users see disparity then cannot reproduce it with TFLM layer-by-layer debugger.
There should no disparities between ANY kernels, optimized or otherwise.

Since TFLite doesn't promise consistency across backends, David's proposal makes sense to me. The goal of the debugger is simply to verify that our ported TFLM reference kernels match TFLite's reference kernels. We aren't trying to match every TFLite mode (e.g., XNNPack or optimized kernels), as that’s impossible. It's not a perfect situation, but it is the reality we are dealing with right now.

closed 2026-10-13 by veblush